apla/brackets-cuwire #5: List of boards is stuck on loading....

lorneb, 2015-02-07:
Suing as Mac,   Have Arduino IDE 1.0.6 which i believe is the last one, and Brackets says it's up to date.
I Installed extension via extension manager option,  the boards drop down just say loading... and thats it.
Port drop down does show list of ports.

anything else I can try?
Thanks

apla, 2015-02-07:
This is a bug, list must be populated or dialog with runtime selection must pop up. Thank you!

But:

actually, brackets-cuwire supports only Arduino IDE 1.5.8+. This is due various reasons:

1. Arduino 1.5 and later have support for various 3rd party hardware;
1. Arduino team is [not accepting patches](https://github.com/arduino/Arduino/pull/2523) for Arduino 1.0.x

To make brackets-cuwire work for you, you can download and install Arduino 1.5.8 and put it into separate location. Then, click on a `gear` icon and paste Arduino 1.5.8 folder location into appropriate input. Click `OK`.

matjes-, 2015-02-08:
Hi,

I have the same problem, I can't populate the boards drop down and also can't compile the text.
Running Windows 7 (64bit), installed Arduino 1.5.8 and add the the location into the cuwire setting. I tried to reinstall everything as well but no change.

thanks

apla, 2015-02-08:
@matjes- , I've fixed obvious things and will release update in few days. can you copy yours console messages to me?

Those messages is available via `Debug > Show Developer Tools` menu, then `Console` tab. Thank you!

matjes-, 2015-02-08:
Thanks, looking forward to it.

Here are the error messages in the console:

 [NodeDomain] Error loading domain "clang": Unable to load one of the modules: C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/completion/../node/clang-domain.js, reason: Cannot find module 'libclang'/utils/NodeDomain.js:148 (anonymous function)
main.js:556 Uncaught TypeError: Cannot read property '0' of undefined
NodeDebugUtils.js:117 [node-log 10:01:36 PM] bad config for: recipe.size.regex.data ^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
NodeDebugUtils.js:117 [node-log 10:01:36 PM] bad config for: recipe.size.regex.eeprom ^(?:\.eeprom)\s+([0-9]+).*
main.js:104 [brackets-cuwire-node] Available ports: Array[0]

mochachocha1981, 2015-02-09:
I'm also getting the "clang" error message:

[NodeDomain] Error loading domain "clang": Unable to load one of the modules: C:/Users/JArnold/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/completion/../node/clang-domain.js, reason: Cannot find module 'libclang'

apla, 2015-02-09:
@mochachocha1981  `clang` issue is different. I tried to publish brackets plugin as soon as possible and added clang loading by mistake. `clang` is planned for next big release

apla, 2015-02-10:
@matjes- @lorneb just released update. please let me know if update fixed issue for you

matjes-, 2015-02-10:
Hi,
it still doesn't work:

NodeSocketTransport - start
/C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/main.js:108 [brackets-cuwire-node] Available ports: Array[1]
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:29:26 PM] bad config for: recipe.size.regex.data ^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:29:26 PM] bad config for: recipe.size.regex.eeprom ^(?:\.eeprom)\s+([0-9]+).*
/C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/main.js:787 Object

apla, 2015-02-12:
I just uploaded debug plugin version to check what's going on

Please uninstall old version, download test release, then install via Extension Manager, button "drag .zip here or install via url" at panel footer.

https://github.com/apla/brackets-cuwire/releases/tag/v%2F0.5.3-test1

Thank you!

matjes-, 2015-02-13:
Here are the console messages:

Download the React DevTools for a better development experience: http://fb.me/react-devtools
/LiveDevelopment/MultiBrowserImpl/transports/NodeSocketTransport.js:87 NodeSocketTransport - start
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 1 version C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 1 hardware C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 1 libraries C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 2 version C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 2 hardware C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 2 libraries C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 1 version C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 1 libraries C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug C:\Program Files\Arduino version: 1.5.8 modern: true
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 0 version C:\Program Files\Arduino
/C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/main.js:104 [brackets-cuwire-node] Available ports: Array[0]
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 2 libraries C:\Program Files\Arduino\hardware\arduino\sam\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait++ 3 libraries C:\Program Files\Arduino\hardware\arduino\avr\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 2 libraries C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] bad config for: recipe.size.regex.data ^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] bad config for: recipe.size.regex.eeprom ^(?:\.eeprom)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 1 hardware C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 1 libraries C:\Program Files\Arduino\hardware\arduino\sam\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:10 PM] debug ioWait-- 0 libraries C:\Program Files\Arduino\hardware\arduino\avr\libraries
/C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/main.js:325 arduinoIDE checked type: null checked: false xx true
/C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/main.js:325 verbose checked type: Array[1] checked: false xx false
/C:/Users/matjes/AppData/Roaming/Brackets/extensions/user/brackets-cuwire/main.js:807 Object
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait++ 1 version C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait++ 2 hardware C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait++ 1 libraries C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait++ 2 version C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait++ 3 hardware C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait++ 2 libraries C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait-- 1 version C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug C:\Program Files\Arduino version: 1.5.8 modern: true
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait-- 0 version C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:16 PM] debug ioWait-- 1 libraries C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] debug ioWait++ 2 libraries C:\Program Files\Arduino\hardware\arduino\sam\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] debug ioWait++ 3 libraries C:\Program Files\Arduino\hardware\arduino\avr\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] debug ioWait-- 2 libraries C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] bad config for: recipe.size.regex.data ^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] bad config for: recipe.size.regex.eeprom ^(?:\.eeprom)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] debug ioWait-- 2 hardware C:\Program Files\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] debug ioWait-- 1 libraries C:\Program Files\Arduino\hardware\arduino\sam\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 6:53:17 PM] debug ioWait-- 0 libraries C:\Program Files\Arduino\hardware\arduino\avr\libraries

apla, 2015-02-14:
Sorry for the comment edit, now it contains only relevant info.

Seems cuwire chokes at `%HOME%\Documents\Arduino\hardware` folder. I'll try to repeat bug on my test machine.

matjes-, 2015-02-14:
Hi,
that make sense. Under windows the hardware folder in your "home" folder is empty by default. You can add new hardware there like "Sparkfun" controller. I copied the content of the hardware folder from the main installation into the "home" location and it populates the dropdown.
I tried to compile the "blink" example from Arduino but a "failed" pops up. Not sure what this is about.

 [node-log 12:53:13 PM] bad config for: recipe.size.regex.data ^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] bad config for: recipe.size.regex.eeprom ^(?:\.eeprom)\s+([0-9]+).*
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] debug ioWait-- 0 hardware C:\Users\matjes\Documents\Arduino
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] debug ioWait-- 3 libraries C:\Program Files\Arduino\hardware\arduino\sam\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] debug ioWait-- 2 libraries C:\Users\matjes\Documents\Arduino\hardware\arduino\sam\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] debug ioWait-- 1 libraries C:\Program Files\Arduino\hardware\arduino\avr\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] debug ioWait-- 0 libraries C:\Users\matjes\Documents\Arduino\hardware\arduino\avr\libraries
extensions/default/DebugCommands/NodeDebugUtils.js:117 [node-log 12:53:13 PM] debug folders information { 'C:\\Program Files\\Arduino': 
   { platform: { 'arduino:sam': true, 'arduino:avr': true },
     boards: { 'arduino:sam': true, 'arduino:avr': true },
     programmers: { 'arduino:sam': true, 'arduino:avr': true },
     runtime: '1.5.8',
     modern: true },
  'C:\\Users\\matjes\\Documents\\Arduino': 
   { platform: { 'arduino:sam': true, 'arduino:avr': true },
     boards: { 'arduino:sam': true, 'arduino:avr': true },
     programmers: { 'arduino:sam': true, 'arduino:avr': true } } }

apla, 2015-03-08:
Just released new version, 0.5.3. Please update plugin using Brackets Extension Dialog. CLI is also updated to version 0.5.9.

Please check if issue persists.

JayPalm, 2015-04-29:
[image]
I think I am having a similar problem. I am using cuwire 0.5.3 and Arduino 1.6.2.
When I click the drop down for the board, I get a blank list as shown above.

LockonS, 2015-09-04:
Using a mac running OS X 10.10.5, and bracket version is v1.4, still got stuck on loading boards. Looking forward to see your work with a fixed plugin.