apla/brackets-cuwire #6: no offline installation method provided

naikrovek, 2015-02-09:
Brackets doesn't honor HTTP_PROXY env vars, and thusly there is no way to install this extension when behind an http proxy.

This is not specifically a cuwire issue, obviously, so please provide an offline installation method.

Alternately, and not ideally, instructions on how to package this extension into a .zip that Brackets can use is sufficient.

apla, 2015-02-09:
All brackets plugin archives is located at [brackets plugin registry](https://brackets-registry.aboutweb.com). I will add instructions to documentation and archives to the releases page.

Thank you!

apla, 2015-02-11:
updated docs and added new release to github releases page

closed 2015-02-11 by apla