apla/brackets-cuwire #7: Cannot set Arduino IDE Location

jockm, 2015-02-11:
On OSX Yosemite, installed Brackets, then Curwire plugin.  Have Arduino 1.5.8 installed at /Applications/Arduino.

Every time I set the Arduino IDE location, I get the following message:

> Runtime issue: 
> No runtimes found! Please install Arduino 1.5.x then relaunch Brackets

I have tried entering:

*  /Applications/Arduino
*  /Applications/Arduino/
*  /Applications/Arduino.app
*  /Applications/Arduino.app/

And all produce the same error.

apla, 2015-02-11:
Just added button to select Arduino IDE. Will release update within next days.

To fix it now, please open Terminal.app and launch command to find all Arduino IDE locations.

```
find /Applications -name Arduino*app
```

Try to insert appropriate location in preferences pane. Please reply does this help or not.

Thank you!

jockm, 2015-02-11:
I determined the location of the Arduino install I wanted to use by opening the terminal and typing `/Applications/Ard`

and then using tab expansion to complete the name.  I then copy and pasted that into the text box.  So essentially the same thing as the list I provided before — none of which worked.

But just in case I used find, and copy and pasted `/Applications//Arduino.app` into the text box, and had the exact same result as before.  As a final "just in case" measure I rebooted my computer and tried again, with the same results.

The only thing I think I failed to mention is that I have multiple Arduino installs, different versions, customized versions, etc; but I fail to see how this could be an issue since Cuwire needs the exact location of the version I want to use.

apla, 2015-02-11:
Ok, that's definitely cuwire error. Do you have `node.js` installed? Brackets plugin shares code with [command line utility `cuwire`](http://apla.github.io/cuwire/cli/) written on `node.js` and it is much easier to detect error using that utility. I will create debug branch and send you instructions how to install it when it's ready.

jockm, 2015-02-11:
I do have node.js installed and I installed the curwire npm as well.  As soon as you have something for me, I am at your disposal to test it.

apla, 2015-02-11:
Try to install latest commit from git repo, globally (with `-g`) or locally — as you wish

```
npm -g install apla/cuwire
```

then run (if installed globally):

```
cuwire --debug boards
```

if installed locally:

```
NODE_PATH=node_modules:$NODE_PATH node cli.js --debug boards
```

apla, 2015-02-11:
please use `-a <Arduino IDE.app>` to define which Arduino IDE installation to use

jockm, 2015-02-11:
First a possibly important point.  I could not run the command

```
npm -g install apla/cuwire
```

Until I upgraded my version of node to the latest.  So apparently I was out of date.  I don't know if this is a factor, but it seemed important to call it out.

Once did upgrade node, here is the result of ` --debug boards`
```
IronGiant:~ jock$ cuwire -a /Applications/Arduino.app/ --debug boards
debug ioWait++ 1 version /Applications/Arduino.app/Contents/Java
debug ioWait++ 1 hardware /Applications/Arduino.app/Contents/Java
debug ioWait++ 1 libraries /Applications/Arduino.app/Contents/Java
debug ioWait++ 2 version /Applications/Arduino.app/Contents/Resources/Java
debug ioWait++ 2 hardware /Applications/Arduino.app/Contents/Resources/Java
debug ioWait++ 2 libraries /Applications/Arduino.app/Contents/Resources/Java
debug ioWait++ 3 version /Users/jock/Documents/Arduino
debug ioWait++ 3 hardware /Users/jock/Documents/Arduino
debug ioWait++ 3 libraries /Users/jock/Documents/Arduino
debug ioWait-- 2 version /Applications/Arduino.app/Contents/Java
debug ioWait-- 2 hardware /Applications/Arduino.app/Contents/Java
debug ioWait-- 2 libraries /Applications/Arduino.app/Contents/Java
debug ioWait-- 1 version /Users/jock/Documents/Arduino
debug ioWait-- 1 hardware /Users/jock/Documents/Arduino
debug /Applications/Arduino.app/Contents/Resources/Java version: 1.5.8 modern: true
debug ioWait-- 0 version /Applications/Arduino.app/Contents/Resources/Java
debug ioWait-- 1 libraries /Users/jock/Documents/Arduino
debug ioWait++ 2 libraries /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/libraries
debug ioWait++ 3 libraries /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/libraries
debug ioWait-- 2 libraries /Applications/Arduino.app/Contents/Resources/Java
bad config for: recipe.size.regex.data ^(?:\.data|\.bss|\.noinit)\s+([0-9]+).*
bad config for: recipe.size.regex.eeprom ^(?:\.eeprom)\s+([0-9]+).*
debug ioWait-- 0 hardware /Applications/Arduino.app/Contents/Resources/Java
debug ioWait-- 1 libraries /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/libraries
debug ioWait-- 0 libraries /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/libraries
cuwire using runtime from /Applications/Arduino.app/Contents/Resources/Java
cuwire boards available:
Arduino AVR Boards (arduino:avr) 1.5.8
LilyPad Arduino USB (LilyPadUSB)
Arduino NG or older (atmegang, models: cpu:atmega168 [ATmega168], cpu:atmega8 [ATmega8])
Arduino BT (bt, models: cpu:atmega328 [ATmega328], cpu:atmega168 [ATmega168])
Arduino Duemilanove or Diecimila (diecimila, models: cpu:atmega328 [ATmega328], cpu:atmega168 [ATmega168])
Arduino Esplora (esplora)
Arduino Ethernet (ethernet)
Arduino Fio (fio)
Arduino Leonardo (leonardo)
LilyPad Arduino (lilypad, models: cpu:atmega328 [ATmega328], cpu:atmega168 [ATmega168])
Arduino Mega or Mega 2560 (mega, models: cpu:atmega2560 [ATmega2560 (Mega 2560)], cpu:atmega1280 [ATmega1280])
Arduino Mega ADK (megaADK)
Arduino Micro (micro)
Arduino Mini (mini, models: cpu:atmega328 [ATmega328], cpu:atmega168 [ATmega168])
Arduino Nano (nano, models: cpu:atmega328 [ATmega328], cpu:atmega168 [ATmega168])
Arduino Pro or Pro Mini (pro, models: cpu:16MHzatmega328 [ATmega328 (5V, 16 MHz)], cpu:8MHzatmega328 [ATmega328 (3.3V, 8 MHz)], cpu:16MHzatmega168 [ATmega168 (5V, 16 MHz)], cpu:8MHzatmega168 [ATmega168 (3.3V, 8 MHz)])
Arduino Robot Control (robotControl)
Arduino Robot Motor (robotMotor)
Arduino Uno (uno)
Arduino Yún (yun)
Arduino ARM (32-bits) Boards (arduino:sam) 1.5.8
Arduino Due (Native USB Port) (arduino_due_x)
Arduino Due (Programming Port) (arduino_due_x_dbg)
debug folders information { '/Applications/Arduino.app/Contents/Java': 
   { platform: {},
     boards: {},
     programmers: {},
     hardware: { error: 'ENOENT' },
     libraries: { error: 'ENOENT' } },
  '/Applications/Arduino.app/Contents/Resources/Java': 
   { platform: { 'arduino:sam': true, 'arduino:avr': true },
     boards: { 'arduino:sam': true, 'arduino:avr': true },
     programmers: { 'arduino:sam': true, 'arduino:avr': true },
     runtime: '1.5.8',
     modern: true },
  '/Users/jock/Documents/Arduino': 
   { platform: {},
     boards: {},
     programmers: {},
     hardware: { error: 'ENOENT' } } }
```

I hope this helps

apla, 2015-02-12:
cuwire cli is working fine… I just uploaded debug plugin version to check what's going on

Please uninstall old version, download test release, then install via Extension Manager, button "drag .zip here or install via url" at panel footer.

https://github.com/apla/brackets-cuwire/releases/tag/v%2F0.5.3-test1

jockm, 2015-02-12:
I uninstalled Cuwire from Brackets, and then installed 0.5.3-test1 as you described... and Cuwire seems to be working as described.  At least I was able to open a .ino file, and most of the dropdowns populated correctly — with the exception of Programmers which is empty.

I will try downloading a sketch to see if that works shortly.

Is there any other tests I can perform, or information I can provide?

apla, 2015-02-12:
Great if it works now. Still have no clue what's happens…

Can you copy log? Menu `Debug > Developer Tools`, then `Console` tab.

Thank you!