apla/brackets-cuwire #9: Cannot find sketchbook libraries

pdaoust, 2015-02-12:
Hi there. I just installed Cuwire (the Brackets extension) and I must say it's much nicer looking than the official Arduino IDE. I am, however, having troubles with sketches that use libraries in the sketchbook folder (I guess this is now `~/Arduino/libraries` rather than `~/sketchbook/libraries` like it was in Arduino 1.0.x). Is there any way I can tell Cuwire where to find those libraries when compiling?

apla, 2015-02-12:
Hi, thank you!

Sketchbook location is changed for a reason: Arduino 1.5+ libraries is different. And sometimes totally incompatible with 1.0.

Setting custom location for sketches is supported in Arduino IDE preferences file. I will support reading this file, but priority is low, sorry. I think you also prefer code completion and library management over sketchbook custom location -)

For now, if you want to use your libraries, the simplest solution is create symlink from `~/sketchbook/libraries` to `~/Arduino/libraries`. Or copy libraries folder.

pdaoust, 2015-02-20:
Thanks for the reply -- my libraries are actually in `~/Arduino/libraries` and they still can't be found. Arduino (1.6) can find them, but Cuwire can't. Any ideas?

apla, 2015-02-21:
Arduino default folder for user's libraries is `~/Documents/Arduino/libraries`. If yours library folder is actually `~/Arduino/libraries`, I'll investigate this.

Also I fixed many issues within file search routine, maybe libraries is suffer from this. I'll let you know when update is ready.

pdaoust, 2015-02-23:
Oh, that's odd -- wonder if the difference in default folders is a Linux vs Mac thing. Are you using Mac OS X?

apla, 2015-02-23:
Thank you!

Yes, my main development environment on mac. I have linux and windows vm to test cuwire.

Just launched Arduino IDE on linux vm and found correct locations.

apla, 2015-03-08:
Just released new version, 0.5.3. Please update plugin using Brackets Extension Dialog. CLI is also updated to version 0.5.9.

Please check if issue persists.

pdaoust, 2015-03-10:
Thanks; I'll check it out when I've got some time!