apla/brackets-cuwire #10: Arduino user libraries not found

maziarzamani, 2015-02-15:
I am using Mac OSX and successfully been working within Arduino IDE while using user libraries in ```~/Documents/Arduino/libraries```. Unfortunately i am not able to compile the code within Brackets, as it is not able to find the header files i am including in the code.

The error:

```
Error: Command failed: /var/folders/64/6n0yv9515hz1n6z3bdzcwv6h0000gn/T/Brackets-cuwire-ad0b5375/Brackets.cpp:23:27: fatal error: RTIMUSettings.h: No such file or directory #include "RTIMUSettings.h" ^ compilation terminated. /var/folders/64/6n0yv9515hz1n6z3bdzcwv6h0000gn/T/Brackets-cuwire-ad0b5375/Brackets.cpp:23:27: fatal error: RTIMUSettings.h: No such file or directory #include "RTIMUSettings.h" ^ compilation terminated.
```

apla, 2015-02-21:
Hi!

Sorry for delay in response.

Found that lib. I think Arduino don't follow their own specification to find a lib by header name. Seems like a trivial issue, I'll let you know when I can compile example from there: `https://github.com/richards-tech/RTIMULib-Arduino/tree/master/RTArduLinkIMU` because error is the same.

apla, 2015-03-08:
Just released new version, 0.5.3. Please update plugin using Brackets Extension Dialog. CLI is also updated to version 0.5.9.

Please check if issue persists.