apla/brackets-cuwire #11: Teensy support

Jorgen-VikingGod, 2015-03-26:
it would be very nice to have also teensy support in cuwire.
There is an existing library and definition for arduino:
http://www.pjrc.com/teensy/teensyduino.html

Here is an overview of the pins out:
http://www.pjrc.com/teensy/pinout.html

I don't know how hard it is to add teensy support >> but it would be awesome!!! :+1: 
Thx for sharing this great plugin with us

apla, 2015-03-26:
Have you tried? I added teensy support few weeks ago, but not tested it. Please share your result if you brave enough for test -)

apla, 2015-03-26:
Forgot to mention: cuwire supports Arduino IDE 1.5.x or later. Teensyduino is actually patch for Arduino IDE. So, You will need to install Arduino IDE 1.6.1 (or another latest version), then install patch from [this thread](https://forum.pjrc.com/threads/23733-Native-support-for-Teensy-3-0-with-Arduino-1-5) and try.

Jorgen-VikingGod, 2015-03-26:
good to know; but I have still issues.
Arduino IDE: 1.61 for Java 7+
Tinsyduino: 1.21
Brackets: 1.2 build 1.2.0-15697 (release 8f82e2a97)
working on Mac 10.10.2

Issue:
no teensy loader was shown and also compile is always "Failed"
but in Arduino IDE it works fine - perhaps something with paths and library access?

apla, 2015-03-26:
Do you have any compilation errors? Or status label just changes color?

Jorgen-VikingGod, 2015-04-07:
sorry, was on vacations...
Only the status label changed the color and said "Failed"

Jorgen-VikingGod, 2015-04-08:
[image]

This is what I'm seeing...
Configuration by default:
[image]

On arduino.app there comes up a "teensy.app" for flashing the teensy board.
This was installed by teensyduino...
After installation there are some libraries and hardware definition (for teensy2, 3, LC) and if course the "programmer"

But on cuwire I can not select a programmer - could this be the problem?

thx for your work.
Hope you can fin the error...

apla, 2015-04-08:
I found that bug a week ago and in a few days I'm going to test a new version, and also will give you to try it. I hope your problem will be fixed with it.

P.S.: Seems cuwire looks really bad in dark.

Jorgen-VikingGod, 2015-04-08:
wow, great to hear!
And yes the comboboxes are bad to read.

Really looking forward to your fixes