apla/brackets-cuwire #13: Cant Compile or Download

IanWalker, 2015-05-02:
This looks like a stunning replacement for the clunky Arduino. Unfortunately I cant seem to get it working.

OS                      Windows 7 home Premium 64 Bit
Brackets             1.3.0-16022
CuWire               0.5.3            
Arduino              1.5.8 (Initially tried with 1.6.3)

CuWire plugin installs OK.
Board drop-down gets populated and I can select the relevant board.
Serial port gets populated OK. Serial terminal works from one user account but not from another.

Programmer drop down does not populate with any options.
Compile and upload buttons give me a Failed status message.

Would really like to get this working - Any advice appreciated.

PS - Is the Arduino 1.6.3 supported ?

IanWalker, 2015-05-03:
Hi - Ok I have loaded the debug version as suggested for another issue.
It seems to get stuck looking for something in the Documents/Arduino folder
<
 [node-log 8:12:48 PM] debug folders information { 'C:\\Program Files (x86)\\Arduino': 
   { platform: { 'arduino:sam': true, 'arduino:avr': true },
     boards: { 'arduino:sam': true, 'arduino:avr': true },
     programmers: { 'arduino:sam': true, 'arduino:avr': true },
     runtime: '1.5.8',
     modern: true },
  'C:\\Users\\Ian\\Documents\\Arduino': 
   { platform: { 'arduino:sam': true, 'arduino:avr': true },
     boards: { 'arduino:sam': true, 'arduino:avr': true },
     programmers: { 'arduino:sam': true, 'arduino:avr': true } } }
>

gweep, 2015-05-13:
Same problem here.    I do keep the Arduino IDE in a non-standard location.  I have successfully pointed cuwire to it.   Perhaps Arduino IDE 1.6.3 is the problem?