apla/brackets-cuwire #15: Brackets-cuwire always loading.

efares, 2015-10-03:
I'm using:
Mac OS X Yosemite
Brackets 1.4.0
Arduino IDE 1.6.5
Arduino Uno board

When I open the Brackets extension I get this screen and it stays like that even after connecting my Arduino Uno and verifying its connection with the Arduino IDE.

Any idea why this is happening?

[image]

dobrMAN, 2015-11-13:
Confirm some problem in Windows 7 x64

Thelmos, 2016-02-13:
I have same problem.

I think this promising tool is being abandoned. User has another repositories with recent activity but this  has no activity from a year now.

phoenixcomm, 2017-05-11:
I just installed this and when I try to pick a board "Loading"
Linux Mint 18
Arduino IDE 1.6.5

angrypig7, 2017-11-24:
Me too.
Loading never seems to end.

DrakoPD, 2018-01-01:
I open the error console, and the problem is with a file called "serialport.node"

`[NodeDomain] Error loading domain "cuwire": Unable to load one of the modules: C:/Users/.../AppData/Roaming/Brackets/extensions/user/brackets-cuwire/node/cuwireDomain.js, reason: Could not locate the bindings file. Tried:

 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\build\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\build\Debug\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\build\Release\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\out\Debug\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\Debug\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\out\Release\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\Release\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\build\default\serialport.node
 → C:\Users\...\AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\compiled\6.11.0\win32\ia32\serialport.node`

I try to install NodeJS, but doesn't work.
Copy the file "serialport.node" from node to the path of extension, I get other error:

`[NodeDomain] Error loading domain "cuwire": Unable to load one of the modules: C:/Users/.../AppData/Roaming/Brackets/extensions/user/brackets-cuwire/node/cuwireDomain.js, reason: %1 no es una aplicación Win32 válida.
\\?\C:\Users\...AppData\Roaming\Brackets\extensions\user\brackets-cuwire\node\node_modules\serialport\build\Release\serialport.node`

If I try to open the file "serialport.node" with brackets, I get a lot of simbols and numbers.

I looking for information, the ".node" file need an special command for use. Probably, in an update of brackets change the form of use this.